sshane/openpilot #134: Stuck at loading screen after following Quick Installation guide

ekawahyu, 2020-07-30:
I followed the quick installation guide and rebooted. But it gets stuck at Comma logo and the loading bar. If I put the stock openpilot back, it loads the dashboard just fine. How to get your fork to work? I have EON with white panda.

thelema, 2020-07-30:
I'm seeing a new problem with a recent update that wasn't there 2 days ago when I drove last.  The problem manifests itself with openpilot dying on startup with a "ValueError":
```
Manager failed to start
Traceback (most recent call last):
  File "./manager.py", line 599, in <module>
    main()
  File "./manager.py", line 576, in main
    manager_prepare(spinner)
  File "./manager.py", line 518, in manager_prepare
    prepare_managed_process(p)
  File "./manager.py", line 332, in prepare_managed_process
    importlib.import_module(proc)
  File "/data/data/com.termux/files/usr/lib/python3.7/importlib/__init__.py", li
ne 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 13, in <module>
    from selfdrive.car.car_helpers import get_car, get_startup_event
  File "/data/openpilot/selfdrive/car/car_helpers.py", line 71, in <module>
    interfaces = load_interfaces(interface_names)
  File "/data/openpilot/selfdrive/car/car_helpers.py", line 43, in load_interfac
es
    CarController = __import__(path + '.carcontroller', fromlist=['CarController
']).CarController
ValueError: source code string cannot contain null bytes
```

I don't see any null bytes in `car_helpers.py:43`, and it looks like this shouldn't have been modified since openpilot 0.5.13, a long time ago.  Similar for the next couple code locations in the backtrace, so maybe the problem is with the file being loaded and not with the code doing the loading.  Maybe something got corrupted in the compilation cache?  I hope this isn't the cause as it is probably a different problem than what you're seeing, but I'm sharing this here as I don't know.   @ShaneSmiskol Thanks for any help you can provide.

d412k5t412, 2020-07-30:
Don't use that script to install it. Use workbench, choose install
openpilot, enter GitHub info for Shane's fork.

ShaneSmiskol, 2020-07-30:
If both of you are experiencing `source code string cannot contain null bytes`, that's caused by either the files or the git blobs being corrupted due to shutting off the EON/C2 unexpectedly (or even randomly at times). I haven't found the reason it occurs randomly, but I know that if the battery dies on the EON or you run `reboot` over ssh often it's more likely to corrupt the folder (again, not sure why). It helps to completely quit openpilot by `tmux a` then ctrl+c before running the reboot command, but not always.

The only fix for me when it happens is to completely reclone the fork as when it corrupts it usually corrupts git blobs as well so you can't simply reset to origin with git. Let me know if recloning helps at all

thelema, 2020-07-30:
Thanks, trying recloning now.  You're right that I've been doing `reboot` over ssh, not knowing it can cause problems.  I'll be careful in the future.  Thanks for the help!

ShaneSmiskol, 2020-07-30:
@thelema there's a tool for NEOS called [emu](https://github.com/emu-sh/.oh-my-comma) that has a reboot command that calls the Android intent to reboot, so in theory it should be the same as rebooting from holding the power button and pressing reboot.  I've been using that for a while without any adverse effects. (you can run it once installing with either of these commands: `shutdown -r` or `emu device reboot`)

ekawahyu, 2020-07-30:
@ShaneSmiskol I have emu installed and switching to commaai fork works just fine, but switching to your fork won't pass the loading screen at all. What else I can debug to fix this issue?

ShaneSmiskol, 2020-07-30:
@ekawahyu Are you getting the null bytes error or is it something else? Also what did you use at first to install my fork, the git command or emu?

ekawahyu, 2020-07-31:
Here is what I did:
1. I did clean NEOS installation and on the setup menu I installed OP 0.7.7 ---> successfully installed, but not working due to no white panda support.
2. I did backup the original OP: `mv openpilot openpilot.org`
3. I installed emu, this cloned the whole commaai repo in.
4. I installed your fork: `emu fork switch shanesmiskol`, it cloned this repo and switched over to `stock_addition`
5. Rebooted the system: `emu device reboot`
6. Got stuck at comma logo with infinite progress bar.
7. If I switch back to commaai using emu, the dashboard worked just fine (0.7.7 release)

Now, where can I see the log for the null bytes error? I am familiar with linux and services but not so much with Android.

thelema, 2020-07-31:
@ShaneSmiskol I used emu to install your fork, I didn't realize it could do reboots properly.  According to this[1], `reboot` internally does a `shutdown -r`, so I'm surprised there's a difference there.

@ekawahyu  Sorry for hijacking your bug report, I believe I had a different problem than you did; I've never seen the "infinite progress bar", although I did let the fork run for a long time after the first reboot.  I didn't realize this, but it's actually recompiling a number of the components on first reboot.  Try having *lots* of patience, i.e. wait 30 minutes for it to finish the reboot.

[1] https://unix.stackexchange.com/questions/64280/what-is-the-difference-between-reboot-init-6-and-shutdown-r-now

ShaneSmiskol, 2020-07-31:
@thelema correct, with emu installed though it overrides the shutdown command in linux (not even available on NEOS) and sends the Android intent to reboot, so it becomes same thing as `emu device reboot`, just quicker to type

ShaneSmiskol, 2020-07-31:
@ekawahyu Oh just read the infinite progress bar part, if you ssh in and type `tmux a` you can see exactly what it's doing or if there's an error

ekawahyu, 2020-07-31:
@thelema hmmm... will try your suggestion to wait, but I am curious about to see if something really is going on, like compiling stuff as you said
@ShaneSmiskol will be back soon after trying `tmux`

ShaneSmiskol, 2020-07-31:
Don't forget the `a` after, just `tmux` will launch a new session and won't allow you to attach to openpilot to see what's happening

ekawahyu, 2020-08-03:
@ShaneSmiskol `tmux a` shows me lots of thing going on! So @thelema was right, the progress bar was stuck because it was actually compiling stuff. The only thing I need to test is the white panda support.

ekawahyu, 2020-08-03:
It seems to work now, closing this issue.

closed 2020-08-03 by ekawahyu

ShaneSmiskol, 2020-08-03:
Glad to hear it! I'll work on white panda support next week