sshane/openpilot #164: Dynamic camera offset causing significant ping-pong

brittonx, 2020-08-12:
On a two-lane road, the car (2018 Prius Prime Advanced), would move over when oncoming traffic was there as expected.  There was a lot of oncoming traffic with significant gaps between the groups of cars.  When the car should have returned to the mid-lane path, it would over compensate then would bounce back and forth until there was no oncoming traffic.
I believe this is a segment showing it:  2eacef693c0640f9|2020-08-11--16-40-07--6

ShaneSmiskol, 2020-08-19:
I don't usually drive on two lane roads much, but I definitely noticed this on a drive I had earlier today. How would reducing the amount it moves over and increasing the time sound to help fix this?

brittonx, 2020-08-19:
First, thank you for this fork. The follow distance and lane change without the wheel tug are great!   I like your suggestion for a change to the behavior. Certainly worth trying.

ShaneSmiskol, 2020-08-25:
Were you able to test the latest change to see if it helped at all?

brittonx, 2020-08-26:
I will be able to test it this afternoon and will let you know tonight.

brittonx, 2020-08-27:
The behavior was better.  That said, I think there may be an issue with the way this is implemented.  If I understand correctly, you're just tweaking the camera offset in real time.  My concern over that is what impact does that have to the ongoing re-calibration that OP does.  When I was testing this latest version, it worked well for a little bit but then the ping ponging started to get worse.  While I don't have data to back it up, it "felt" like some adjustments were happening in the background.

ShaneSmiskol, 2020-09-10:
It's possible the live angle offset learner is affecting how the offsetting is working, however Dynamic camera offset uses a PI controller so it should be able to adjust in real time for slight movements in the lane. I am aware of a bug where it jumps back to center after keeping rather than smoothly moving over, I'll look into that soon

closed 2020-09-11 by ShaneSmiskol