sshane/openpilot #171: Feature Request

krkeegan, 2020-08-20:
Shane, any interest in adjusting the amount of time that the C2 remains on after the car is turned off?  I have previously used Erich's suggested changes to [board.cc]( https://github.com/ErichMoraga/openpilot/commit/180a976266faff336c92f8ff04859bec76918972) which change the time to 5 hours and move the battery voltages up by about .5 a volt?

I ask, because I just killed the battery on my Rav4 hybrid.  Part of it was my fault, I moved the car a few times over the last week without letting it run for very long.

Anyways, I completely understand if this isn't what you want in you fork, I can go back to making the changes manually.

brittonx, 2020-08-20:
Perhaps this needs to be a setting.  I had submitted an issue because the shorter timeout that was in this fork recently was a major annoyance compared to stock OP.  I never had a problem with stock OP over draining the 12V battery in my Prius Prime.  I don't want to have to wait for the C2 to boot every morning.

brittonx, 2022-03-30:
> @krkeegan are you satisfied with the solution? My Toyota Corolla 2021 Hybrid died 4 times in 12 months already.
> 
> I asked on chat and received the link to the `MAX_TIME_OFFROAD_S` [ErichMoraga@c0ec38a](https://github.com/ErichMoraga/openpilot/commit/c0ec38a62116fc0467147ede7c7419fc071ce3f6) I suppose it would be nice to make it configurable with opEdit. @sshane would you accept such a PR?

The timeout is configurable in op_edit. The Parameter is called disable_charging.

Namek, 2022-03-30:
I just found out there is `disable_charging` on the config params list: https://github.com/sshane/openpilot/blob/SA-master/common/op_params.py#L101 <3