sshane/openpilot #181: vw j533 integration

cdennis12280, 2020-08-26:
Hi this fork doesnt work on VAG group cars that have their panda integrated at J533, any chance you could merge the code from Dragon pilot or Jyoung Repo that enables J533 integration for VAG group cars. 

looking fwd to trying your fork soon.

ShaneSmiskol, 2020-08-26:
Hey do you have a specific commit or commits I need to merge to make this work for you?

cdennis12280, 2020-08-27:
I'm not sure which files are typically modified for new car support but it would be those ones.
Anyones related to volkswagen i.e.

panda/tests/safety/test_volkswagen_mqb.py
selfdrive/car/volkswagen/carcontroller.py
selfdrive/car/volkswagen/carstate.py
Etc etc

This may help?...

https://github.com/ShaneSmiskol/openpilot/compare/stock_additions...jyoung8607:vw-community-devel#diff-dab0972aae396b470beede3d3f1a23cf

closed 2020-09-10 by ShaneSmiskol