sshane/openpilot #182: Manager failed to start - Syntax errors in values.py

brittonx, 2020-08-26:
I came out to the car today and found this error.  It comes back on every reboot.
[image]

I edited the file, there was a missing comma at the end of the line before that.

Then, on reboot, I got the following error:
[image]

Not sure what is messing up the values.py

brittonx, 2020-08-26:
I removed lines 143-153 on my C2 then it was able to boot.

brittonx, 2020-08-26:
I think a problem is if this gets hosed, I don't thing the updater is able to update the file with a revised version where you may have already corrected the issue.  You may want to have a "stock_additions_staging" where you can test that updates don't prevent the unit from booting before the updates get pushed out to those of us who are running this fork.  This is the second time this has happened.  The first time, I just did a reinstall which is a pain.

ShaneSmiskol, 2020-08-26:
Hey @brittonx, this issue was fixed about an hour after the bad PR was merged, but if your C2 was on when the bad commit went through it pulled that and stopped manager from updating itself to get the latest fixing commit. You're right about the updater, if manager doesn't start it can't get updates.

I'll add a git pull line in the openpilot_launcher file so it can get new updates on boot up if there's an error, that will be before openpilot even starts up.

Sorry about this, I failed to check that a PR had correct syntax before merging a new fingerprint. If you can ssh in all you need to do is `git pull` and it should be fixed now. Make sure to `git reset --hard origin/stock_additions` if you changed stuff locally so git pull will go through

brittonx, 2020-08-27:
Thank you for the update. I'm okay if you want to close this issue.

ShaneSmiskol, 2020-08-27:
Merged into SA, if this ever happens again ensure your device is on WiFi and tap the new Reset button to perform a git pull and reset --hard!

closed 2020-08-27 by ShaneSmiskol