sshane/openpilot #206: SUGGESTION: Add a Stock_Additions build number to the OP version number

brittonx, 2020-09-10:
Several updates have come in to the C2 but it still just shows 0.7.7 release.  It would be helpful to have a build number included. For example: 0.7.7.42

TroyTanon, 2020-09-10:
I think the minor updates don’t get added just due to the extra work. Most updates are minor and can be viewed in the comment logs.

ShaneSmiskol, 2020-09-10:
I used to have Stock Additions 0.7.4 (v1.1 for ex.) in my changelog, but thought that I didn't add too much in each version to justify it. But I think now it might be a good idea since I'm adding much more features than when I first started. For now you can check this file out for the SA changes for openpilot version: https://github.com/ShaneSmiskol/openpilot/blob/stock_additions/SA_RELEASES.md

closed 2020-09-10 by ShaneSmiskol