sshane/openpilot #275: Feature Request: Brake Light Indicator

TroyTanon, 2020-11-10:
Display light when car begins braking.

Jordi-m, 2020-11-12:
For those of you who want to try to add this feature, this PR from the kegman fork might help you to find the correct pieces of code to do so: https://github.com/kegman/openpilot/pull/10/files

Jordi-m, 2021-01-17:
@TroyTanon FYI: This feature request has been created and merged by @ShaneSmiskol yesterday: https://github.com/ShaneSmiskol/openpilot/pull/309 (Thanks Shane, I also really appreciate it)

TroyTanon, 2021-01-17:
Yep. I've tested it a little. Does it work while comma is engaged? I didn't see it trigger but I was 100% sure if the vehicle was braking.

Jordi-m, 2021-01-17:
I'll give it a try in my Toyota Corolla when I have my C2 replacement unit back :-) (probably 2-3 weeks from now) Testing it shouldn't be that hard right, should already be noticable at traffic lights with a red light.

Do you drive a Toyota? It could be that this feature is restricted to cars that share this information via the CAN bus (I don't know if all OP supported cars send out their brake light data - car_state.getBrakeLights() - https://github.com/ShaneSmiskol/openpilot/pull/309/files#diff-f3f92e85a5a0f6b494baa6bf68ceb60e36907f816f323a97785588b441f45a04R366)?

TroyTanon, 2021-01-17:
Didn't think about testing it like that. I've only driven on freeway since the merge. Yes I drive a 2020 Corolla.

ShaneSmiskol, 2021-01-18:
brakeLights never worked for me (as `cp.vl["ESP_CONTROL"]['BRAKE_LIGHTS_ACC']` was always 0 in Cabana unless the user braked) but I thought it was just a bug for my car. I found another signal for Toyota that should work for detecting when either the car is braking or the user is braking, let me try and test it.

ShaneSmiskol, 2021-01-18:
Can somebody test this with a Toyota and let me know how it goes? https://github.com/ShaneSmiskol/openpilot/pull/315 Thanks!

Jordi-m, 2021-01-18:
I'll get back here with testresults (Toyota Corolla 2020) once I have my replacement C2 back.

Thaixican, 2021-01-18:
It works! Also with cruise off and Brake Hold working, the number at top does not stay red I think as intended. It only stays red with brake pedal depressed regardless of cruise mode, and stays red when OP uses the brakes.
Tested on 2020 Lexus RX450h (like _TSS2.0_). Now I can tell if OP braking annoys drivers behind me with too much braking and not minor regen braking (which occurs while brake lights stay off). Praise be @ShaneSmiskol 🙏

TroyTanon, 2021-01-18:
Not showing braking while comma is active/in control.

ShaneSmiskol, 2021-01-20:
Seems to be an issue with the signals available to us, brakeLights only works for the user's pedal and for Toyota there's no other known signal that can give us this information. We can't show it on brake > 0 since the car doesn't brake until a certain threshold, which changes with speed, it's not constant. So for right now this feature only works when the user brakes or with cars that have a good way to detect the brake lights

closed 2021-01-20 by ShaneSmiskol

Jordi-m, 2021-01-30:
I can confirm this is working as expected (nice!) on my Toyota Corolla Touring Sports Hybrid 2020 (EU):

- When I press the break, the speed indicator turns red
- When OP is engaged and breaks by itself, the speed indicator turns red

TroyTanon, 2021-01-30:
I can confirm that it doesn't work with Comma.ai. 2020 Corolla SE. Hybrid likely works due to something happening on the back end to adjust the hybrid system and regen breaks.

TroyTanon, 2021-01-30:
Although would it be possible to use Comma signal for when breaking is starting for the comma side?