sshane/openpilot #283: Openpilot boots to a black screen

mfd00gie, 2020-11-21:
EON old school HTC one. EON updated recently to OP .10 worked briefly. Phone (EON) boots and goes to the comma boot screen. after the comma icon, the screen goes completely black. EON non responsive except for pushing the power button which pops up the shut down and restart options. EON does not communicate with car ('16 civic) white panda and giraffe.

ShaneSmiskol, 2020-11-22:
Hi, can you run `tmux a` and show what it displays when it's in this black state? I would try `git pull` to see if there's any updates you're missing out on from my fork.

ShadowVlican, 2020-11-24:
Mine did that too but a git pull fixed it

mfd00gie, 2020-11-28:
I don't know how to run tmux. I have a mac and a pc, can you do it with that or so you need a Linux based system?

closed 2021-01-03 by ShaneSmiskol

JoelHurd, 2021-01-16:
Getting the same but on an EON gold. `tmux a` spits out the what's posted below. There might be more above but workbench isn't letting me page up on the command line atm.
```running once
Using interface 'wlan0'
OK
3
3
/data/openpilot/launch_chffrplus.sh: line 34: echo: write error: No such device
/data/openpilot/launch_chffrplus.sh: line 63: /proc/irq/184/smp_affinity_list: No such file or directory
/data/openpilot/launch_chffrplus.sh: line 63: /proc/irq/185/smp_affinity_list: No such file or directory
/data/openpilot/launch_chffrplus.sh: line 63: /proc/irq/186/smp_affinity_list: No such file or directory
pid 209's current scheduling policy: SCHED_OTHER
pid 209's current scheduling priority: 0
pid 209's new scheduling policy: SCHED_FIFO
pid 209's new scheduling priority: 52
pid 210's current scheduling policy: SCHED_OTHER
pid 210's current scheduling priority: 0
pid 210's new scheduling policy: SCHED_FIFO
pid 210's new scheduling priority: 52
pid 322's current scheduling policy: SCHED_OTHER
pid 322's current scheduling priority: 0
pid 322's new scheduling policy: SCHED_FIFO
pid 322's new scheduling priority: 52
pid 323's current scheduling policy: SCHED_OTHER
pid 323's current scheduling priority: 0
pid 323's new scheduling policy: SCHED_FIFO
pid 323's new scheduling priority: 52
Result: Parcel(00000000 00000001   '........')
Traceback (most recent call last):
  File "./manager.py", line 18, in <module>
    from common.basedir import BASEDIR
  File "/data/openpilot/common/basedir.py", line 4, in <module>
    from selfdrive.hardware import PC
  File "/data/openpilot/selfdrive/hardware/__init__.py", line 5, in <module>
    from selfdrive.hardware.eon.hardware import Android
  File "/data/openpilot/selfdrive/hardware/eon/hardware.py", line 8, in <module>
    from cereal import log
ImportError: cannot import name 'log' from 'cereal' (unknown location)'

ShaneSmiskol, 2021-01-16:
@JoelHurd That's weird, looks like something with the openpilot install isn't set up correctly since it can't find a common library. I'd try recloning the fork you're using and then reflashing NEOS if that doesn't work.