sshane/openpilot #390: Support for kia hyundai op long?

oricalworks, 2021-04-21:
Hi I have op long working on my kia stinger using this fork and moving my radar onto can bus 1 using a harness is there anyway you can add the kia stinger Long control with scc harness to your fork. Your work on the adaptive cruise distance is really awesome. If you have a zelle id donate
https://github.com/oricalworks/openpilot

ShaneSmiskol, 2021-04-21:
Hi, which of his branches works?

oricalworks, 2021-04-21:
Xpspro

oricalworks, 2021-04-21:
It uses a fork of andy's panda firmware too
https://github.com/oricalworks/panda
Branch xps083

oricalworks, 2021-04-21:
Send me your zelle on discord johnpr#5623 hard work doesn't go unnoticed :)

ShaneSmiskol, 2021-04-23:
Is this all that's required? There seems to be other code not related to long control: https://github.com/oricalworks/openpilot/compare/ab6ada9d8628ac5f84a5a00fba9bbb37247dc471...oricalworks:xpspro

To be honest, this seems quite involved to add and maintain for each new release since I rebuild the fork manually each time. If you can get it down to a single cherry-pickable commit you could do that on top of SA yourself.

oricalworks, 2021-04-24:
Check out https://github.com/andy741217/dragonpilot/tree/shane082_long

oricalworks, 2021-05-08:
Please add HKG support we have radar and vision now.
Here is the cleaned up repository that works https://github.com/oricalworks/op4 branch hkg-nosmdps
Thanks for your time

ShaneSmiskol, 2021-05-08:
Hey, do you have a specific commit range for me to look at? There's quite a bit of changes there. Also, if you could detail what support you want added exactly. Radar support or vision long support?