sshane/openpilot #401: Not to break when gas override is done

chicman, 2021-04-28:
I got a thing that would improve driving alot.  

When gas is overriding and vehicle reaches faster speed than set speed, gas release will cause OP to apply a little break to decrease speed to SET. 

This is different from TOYOTA stock setting, that will let the car just slide/coast.  I do prefer stock behavior for smoother driving. 

FYR
BEST WISHES!

ShaneSmiskol, 2021-04-29:
openpilot already takes a while to brake after overriding, plus there's some code to make the braking less abrupt already. If we don't brake at all, then on some older TSS1 vehicles the car will attempt to slow you down via jerky engine braking.

closed 2021-04-29 by ShaneSmiskol

chicman, 2021-04-29:
On my RAV4 TSS2, break applies almost instantly after gas released. Can we have a op_edit param for (COAST if gas override finish)?

4oSmile, 2021-11-11:
I move to sshane' fork few days ago and it have everything I need:
- standstill hack is AWESOME for stop & go on my car (Prius'17 TSS1 with SmartDSU) 
- distance profiles switch by button on my steering wheel
- change line by blinker lever, even just with small touch (without lever switching)
- dynamic gas is also True in parameters but I not sure if used beacause I don't have comma pedal, only SDSU ??? 
- no disengagement on gas
and the only tiny/little/small unperfection is this issue #401 which applies abrupt brake when gas released Especially when difference between current speed and setted is large! I hope someday something can be made to remove/reduse it IF NO vehicle is infront of course!

I'll be glad to help with feedback and test for all these (and new) features.
Thank you @sshane and good luck :)