sshane/openpilot #402: Cruise Control Malfunction , LCA Malfunction when attempting Engage

hwhorf, 2021-04-28:
I installed the fork and rebooted. EON loads Stock Addition and I get a different GUI but when trying to enable/engage self driving i get a crouse control Malfunction error on my 2020 corolla

ShaneSmiskol, 2021-04-29:
Do you get this on stock openpilot? I don't see any errors about this, check the `/data/community/crashes` dir for a `latest` file. If there's nothing about a malfunction in the file, check some of the other log files.

hwhorf, 2021-04-29:
Thanks. I didn't get this error on stock openpilot before I installed the fork. I tried reinstalling the fork and same error happens (I see the new GUI but ACC doesn't engage). I see "vehicle online" on EON but it says "connect error" in red or "connect online" in yellow. I installed DP and similar error happens. I will check logs next

ShaneSmiskol, 2021-04-29:
Try reflashing panda as well if you get this on other forks as well.

ShaneSmiskol, 2021-05-03:
Is this fixed now on my fork?

closed 2021-05-11 by ShaneSmiskol