sshane/openpilot #403: Request to add finger print for 2021 Highlander

SliderApp, 2021-04-28:
Under: 
CAR.HIGHLANDER_TSS2: {
    (Ecu.eps, 0x7a1, None): [
      b'8965B48241\x00\x00\x00\x00\x00\x00',
      b'8965B48310\x00\x00\x00\x00\x00\x00',
    ],
    (Ecu.esp, 0x7b0, None): [
      b'\x01F15260E051\x00\x00\x00\x00\x00\x00',
      b'\x01F15260E110\x00\x00\x00\x00\x00\x00',
    ],
    (Ecu.engine, 0x700, None): [

Can you please add:

b'\x01896630EB2200\x00\x00\x00\x00',

This is the fingerprint for my car.

chicman, 2021-04-29:
Just Pull request at the values.py

SliderApp, 2021-04-29:
you know I'm not really sure on how to do a pull request a specific file from the master-ci repo to shanes fork, I'm using the github web interface not sure I can do this on it.

chicman, 2021-04-29:
Click edit on top right at the file. GitHub will generate a patch at your account, fill in changes and information, you will be able to submit a pull request right there.

SliderApp, 2021-04-29:
cool thanks I didn't even know I could do that. Thanks for pointing me in the right direction!

closed 2021-04-29 by SliderApp