sshane/openpilot #411: Eliminate grinding/ripping brake sounds on 18 Honda Odyssey with pedal

hooser13, 2021-05-04:


hooser13, 2021-05-12:
I tested it with stock 8.4 and it had the brake sounds as well.  It's a specific problem with the comma pedal.  With 8.4 I also got the creep ahead.

ShaneSmiskol, 2021-05-18:
Try either my fork or stock out now to see if it's gone!

hooser13, 2021-05-18:
I will try your fork in the next couple days.  As you know through Discord, Honda+Pedal=Kegman.  It would be great if other forks or stock OP would interface with Hondas and the pedal better, but I know that the pedal isn't supported by Comma.  This was never a problem without the pedal, but I like the functionality of OP below 23 MPH.  Do you recommend Stock 8.3 or 8.4?

ShaneSmiskol, 2021-05-18:
Stock supports pedal, and this fix Chris tried to upstream but was denied is now in upstream openpilot. Forks just need to cherry pick it

VirtuallyChris, 2021-05-18:
Can close this now :)

hooser13, 2021-05-18:
The problems with honda vehicles using the pedal is the brake system behaves/sounds erratic and abnormal.  There are extra and weirdbraking sounds that aren't heard without the pedal.  I will try the update tomorrow and check on any changes.  Thanks for the help and support.

hooser13, 2021-05-19:
Very impressed with the fork while using the pedal in a 18 Honda Odyssey.  The brake sounds and operations weren't even noticeable with my drives.  I appreciate the changes/updates you have done and consistent management of this fork.  Worked much better than 8.4 which I know is still in release2-staging so not fully released.

closed 2021-05-19 by hooser13