sshane/openpilot #414: use_lqr True causes red screen of panic on 2021 Toyota RAV4 Hybrid

alexsielicki, 2021-05-09:
Comma 2 hardware running Stock Additions Update 3 (0.8.2) fork malfunctions when use_lqr toggle is set to True on 2021 Toyota RAV4 Hybrid. When enabling cruise control, the car's dash beeps and shows "CruiseControl Malfunction" error message and the Comma screen turns red with "TAKE CONTROL IMMEDIATELY - Controls Unresponsive" message showing while panic sound plays loud over the built in speaker.

Setting use_lqr to false returns normal operation. Also, use_lqr true is no problem when rav4TSS2_use_indi is also true, maybe because rav4TSS2_use_indi overrides use_lqr?

See it in action here:
https://youtu.be/2CJgXPmFHU0

ShaneSmiskol, 2021-05-11:
Can you try that out now? It was missing some code

alexsielicki, 2021-05-11:
I can try it later today. I'm new to this, so do I need to ssh to comma and do a git pull? From what branch? Or will it just update itself?

alexsielicki, 2021-05-12:
Never mind, found your fix in the latest commit, comma notified me of the update, I installed it, rebooted, toggled use_lqr, rebooted again, and everything worked this time. Drove about 30 miles with it on and no issues. Thanks for the quick fix.

So does rav4TSS2_use_indi override use_lqr like I thought?

closed 2021-05-12 by alexsielicki

ShaneSmiskol, 2021-05-12:
Nope, LQR always overrides, the way I have it set up.