sshane/openpilot #420: Better adaptability when driving on slight to moderate hills

hooser13, 2021-06-05:
When driving my 18 Honda Odyssey with a Comma Pedal, going downhill the system will brake after going 2-4 MPH over the set speed.  When the vehicle goes to climb the next hill the it will wait to accelerate hard enough until it has dropped at times 5 MPH under the speed limit.  You have a situation where it will brake downhill and lag acceleration going uphill

ShaneSmiskol, 2021-06-05:
I'm not sure why braking is off, since you use the same tune as stock Toyota and send deceleration commands, but acceleration with pedal will be fixed as soon as I have time to fit a function that works at all speeds, and hopefully it will get upstreamed to stock. This isn't an issue with my fork unless you say that there's a difference between stock and here. If you want, you can also open this issue on stock openpilot to get more visibility, and have other users with the same issue come and post information for me to use!

sshane, 2021-12-15:
Closed for inactivity

closed 2021-12-15 by sshane