sshane/openpilot #421: Pre Collision Malfunction on 21' Toyota Highlander after using update button on Comma2 with SA installed

SliderApp, 2021-06-06:
I been using SA for the past few days on a 21' Highlander, it wasn't working prior to the LKA fix that was merged into SA.

I got it working with intermittent issues when turning on the device after starting the car, but if Open Pilot was running and then I started the car it was working fine.

What I noticed was that after that update button was pressed, which I did over cellular, it would always give me this Pre Collisions malfunctions that I was unable to get rid of.

SliderApp, 2021-06-06:
did a test and removed the commit from yesterday: https://github.com/ShaneSmiskol/openpilot/commit/be1d5a4bdf67ca23b4d1feb76ed3df38efe966c5

I got the pre collision malfunction like the prior build but after car shutdown and restart it worked correctly. Not sure why I get that message on first startup of SA but it was like that prior to this above commit.

Something about the commit above does not allow  shutdown/restart to remove malfunction alert to go away.

SliderApp, 2021-06-11:
So SA updated itself again, forgot to turnoff auto update. 

Got the Pre-Collision System Malfunction error, even though I get this error OpenPilot still runs when activated using cruise control buttons. Any idea why this is getting triggered, is there some logs I can get you @ShaneSmiskol ?

[image]

ShaneSmiskol, 2021-06-12:
Hmm, I don't think any of those error could have caused this. Make sure you pull to get the commit that fixed the mistake I made a few days ago. If openpilot starts up, then all should be working. Also if you don't have a DSU or a comma two then you have to wait for openpilot to start before starting the car. Was there any warnings or errors on the actual device (not the car)?

SliderApp, 2021-06-13:
I am using a CommaTwo, I'll try pulling the latest today. 

No errors on device at all, it actually works, just the pre collision message on the car that won't go away. 

My Dongle ID b2e1ad6f5ee4eefe

Let me know if you need me to add you to log access, I just need your email.

SliderApp, 2021-06-13:
So I checked the commits on device and latest commit was  be1d5a4bdf67ca23b4d1feb76ed3df38efe966c5 "clip for safety"

sshane, 2021-08-02:
Hi, are you still experiencing this?

SliderApp, 2021-10-11:
this hasn't been an issue after reinstalling SA a month ago.

closed 2021-10-11 by SliderApp