sshane/openpilot #425: Disable Charging Timer Not Initialized Until Ignition

krkeegan, 2021-06-16:
I have a RAV4 Hybrid with a tiny 12v battery.  I have had a few instances of a completely dead battery, even after setting the `disable_charging` parameter to 8 hours.  To debug things I used a trickle charger and I think I finally caught the issue.  It seems that after a reboot, the `disable_charging` parameter is ignored or the timer is not started.  This is fixed when the ignition goes to on.  If you reboot the C2 with the car off and don't turn the car on, the device will stay on for days maybe forever.

In my case, I believe the issue was caused by automatic updates.  The device must have updated after the car was off, and then rebooted and sat on for days.  If the device is off, and is triggered to turn on when the car starts, everything seems to work fine.

I don't know if this is something that can be fixed in your fork, or if this is a glitch from the comma master.  But you may want to consider adding an extra note to the documentation or the `disable_charging` op_edit param.

sshane, 2021-08-02:
Are you still experiencing this? Try setting the `ForcePowerDown` param to 1 after the device starts and see if it shuts down appropriately

krkeegan, 2021-08-02:
Sadly, I just sold and shipped my C2.  Now I am waiting for my C3 to arrive.  I may wait a bit before putting custom firmware on it just to let things settle out a bit.

You can close this if no one else is reporting the issue.