sshane/openpilot #458: op_edit errors on comma three

brittonx, 2021-08-23:
comma@tici:/data/openpilot$ sudo python op_edit.py
  File "op_edit.py", line 237
    print(msg, flush=True, end='\n' + end)
                    ^
SyntaxError: invalid syntax
comma@tici:/data/openpilot$ ./op_edit.py
Traceback (most recent call last):
  File "./op_edit.py", line 3, in <module>
    from common.op_params import opParams
  File "/data/openpilot/common/op_params.py", line 7, in <module>
    from selfdrive.hardware import TICI
  File "/data/openpilot/selfdrive/hardware/__init__.py", line 5, in <module>
    from selfdrive.hardware.eon.hardware import Android
  File "/data/openpilot/selfdrive/hardware/eon/hardware.py", line 10, in <module>
    from cereal import log
ImportError: cannot import name 'log' from 'cereal' (unknown location)

brittonx, 2021-08-23:
I'm wondering if it is still compiling.  It's been stuck on the screen with the big comma and no progress bar for a while...

brittonx, 2021-08-23:
Does it need to be run with sudo?

sshane, 2021-08-23:
What does `tmux a` say when the logo comes up with no progress bar?

brittonx, 2021-08-23:
Waiting for wayland
wayland ready
adjusting time from '2021-07-21 15:00:39.508255' to '2021-08-23 15:15:00'
Mon 23 Aug 2021 03:15:00 PM UTC
INFO:root:System time valid
/data/openpilot has been modified, skipping overlay update installation
Cloning into '/data/openpilot/cereal'...
fatal: unable to access 'https://github.com/sshane/cereal.git/': Could not resolve host: github.com
fatal: clone of 'https://github.com/sshane/cereal.git' into submodule path '/data/openpilot/cereal' failed
Failed to clone 'cereal'. Retry scheduled
Cloning into '/data/openpilot/laika_repo'...
fatal: unable to access 'https://github.com/commaai/laika.git/': Could not resolve host: github.com
fatal: clone of 'https://github.com/commaai/laika.git' into submodule path '/data/openpilot/laika_repo' failed
Failed to clone 'laika_repo'. Retry scheduled
Cloning into '/data/openpilot/opendbc'...
fatal: unable to access 'https://github.com/commaai/opendbc.git/': Could not resolve host: github.com
fatal: clone of 'https://github.com/commaai/opendbc.git' into submodule path '/data/openpilot/opendbc' failed
Failed to clone 'opendbc'. Retry scheduled
Cloning into '/data/openpilot/panda'...
fatal: unable to access 'https://github.com/commaai/panda.git/': Could not resolve host: github.com
fatal: clone of 'https://github.com/commaai/panda.git' into submodule path '/data/openpilot/panda' failed
Failed to clone 'panda'. Retry scheduled
Cloning into '/data/openpilot/rednose_repo'...
fatal: unable to access 'https://github.com/commaai/rednose.git/': Could not resolve host: github.com
fatal: clone of 'https://github.com/commaai/rednose.git' into submodule path '/data/openpilot/rednose_repo' failed
Failed to clone 'rednose_repo'. Retry scheduled
Cloning into '/data/openpilot/cereal'...
fatal: unable to access 'https://github.com/sshane/cereal.git/': Could not resolve host: github.com
fatal: clone of 'https://github.com/sshane/cereal.git' into submodule path '/data/openpilot/cereal' failed
Failed to clone 'cereal' a second time, aborting
Traceback (most recent call last):
  File "./build.py", line 10, in <module>
    from common.basedir import BASEDIR
  File "/data/openpilot/common/basedir.py", line 4, in <module>
    from selfdrive.hardware import PC
  File "/data/openpilot/selfdrive/hardware/__init__.py", line 5, in <module>
    from selfdrive.hardware.eon.hardware import Android
  File "/data/openpilot/selfdrive/hardware/eon/hardware.py", line 10, in <module>
    from cereal import log
ImportError: cannot import name 'log' from 'cereal' (unknown location)

brittonx, 2021-08-23:
That looks like it is progressing.  Should I reboot after it finishes?

brittonx, 2021-08-23:
That did it.  Based on the tmux, it looks like the wifi had not connected yet when it was first trying to update the submodules.

closed 2021-08-23 by brittonx