sshane/openpilot #478: Put back red speed when braking

VolLearner, 2021-10-03:
In sa_additions we had the speed turn red when braking, In the latest releases of sa_master it dropped out please put it back.

Thank you

SliderApp, 2021-10-11:
I would would like this feature added back to. Super useful especially in downhill scenarios to see if the car is braking to fight gravity when the coast is clear.

Jordi-m, 2022-06-17:
I've been using a static version of Shane's fork for over a year now (never updated), also because of this feature (besides dynamic following profiles). My C2 bootlooped today and I had to reset the whole thing. Now I'm using Shane's latest master.

I've been testing PR #309 beginning 2021 (Toyota Corolla) which did bring the red speed indicator to shane's fork. Liked it a lot. It's been really usefull to add a little speed yourself when you see it's braking to fast/often (e.g. slowly driving stop-go traffic[jam] situations). And in general, it's just good to know when the car is breaking versus not adding gas.

I don't think I've been involved enough to know the latest OP code, but I'll try to take a look at it in the coming week or so. Would really like to have that red indicator back.

If you'd like to have a version of Shane's fork where the red indicator was just added: https://github.com/Jordi-m/openpilot (this is what I've been using for over a year now).