sshane/openpilot #495: Comma 2 - Prius Prime 2020 - Panda not sensed

dhamo1, 2021-11-19:
**Describe the bug**
I just bought Comma 2 and tested for few days with stock release. Then I installed sshane / openpilot but it couldn't sense panda.

**How to reproduce or log data**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

sshane, 2021-11-19:
Hi, try these instructions to flash your panda: https://github.com/commaai/openpilot/wiki/Panda-Flashing

sshane, 2021-12-15:
Did it fix it?

closed 2021-12-21 by sshane