sshane/openpilot #497: Openpilot unavailable low battery

wongjunjin, 2021-12-03:
**Describe the bug**

Ever since I upgrade to 0.8.9, I am unable to engage openpilot as I receive "Openpilot unavailable low battery". I've been receiving this for the last few weeks of drive.

This happens most of the time. I would need to unplug the USB-C and reconnect it so that it will work, but only for a short period before I get an alarm saying "low battery" again. From then on, I would not be able to re-engage openpilot, as I keep receiving the "Openpilot unavailable low battery" issue.

I would need to remove and reconnect the USB-C again for it to work for a few minutes only. Rebooting C2 does not help. Turning off and on my car does not help either.

**Additional context**

This issue has been resolved in the 0.8.10 main branch of open pilot. [https://github.com/commaai/openpilot/commit/74a3c92704b1d6597002d6c5da020ba8ff4393a1](url)

Would appreciate if you could implement the same code in Stock Additions.

sshane, 2021-12-15:
Should be fixed soon when I update my fork (moving `SA-master-acados` branch over to `SA-master`)

sshane, 2021-12-21:
Updated to 0.8.12, should be okay now!

closed 2021-12-21 by sshane

wongjunjin, 2021-12-21:
Thanks!! I’ll test it out!