sshane/openpilot #557: Sentry tripped while car is already in an unlocked state

bensonkwek, 2022-01-23:
To reproduce
- car is in locked state
- unlock car
- enter car , shut the door
- wait for a min
- open door
Result : sentry warning tripped

Hybrid RAV4 2021

Let me know what other information you need

sshane, 2022-01-24:
The problem is I want it to arm after being inactive for a bit as I don't have the lock signals for other cars coded. I can raise the timer to 5 minutes though if that'll help. Right now it's only 2 minutes

Asherlc, 2022-02-28:
Same issue for me, would it be possible to enable the timer only for unsupported cars?

sshane, 2022-03-25:
Another problem is if you start the C3 while the car is already locked, it has no idea what the lock status is, as it's only published when it changes when the car is off. Feel free to make a PR to make it easy to stop the alarm if it's a verified user, or raise the timeout.

closed 2022-03-25 by sshane