sshane/openpilot #560: "Low Memory Reboot Device" error while attempting to disable lane speed alerts.

MattBrownies, 2022-02-06:
### Describe the bug

After driving 45 minutes I was cruising in the slow lane which made the lane speed alerts quite annoying. I went to disable while underway and I received a "Low Memory Reboot Device" error. However, a few minutes after I received the error I was able to reeangage openpilot. Once stopped I went to reboot the device and the error popped up again and didn't return once reboot was complete.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

8a941ca545bbf2ad|2022-02-05--19-02-58--0

### openpilot version

8.13

### Additional info

2021 Rav4 hybrid.
Issue happens around 19:50.
The drive is actively uploading at time of submitting this bug report.

MattBrownies, 2022-02-14:
received a second error after starting the car. Made this drive public and preserved. 8a941ca545bbf2ad|2022-02-13--16-48-06

perkarlsson, 2022-03-20:
I’m having the same issue; 2021 RAV4 hybrid. 

Low memory / reboot — lane alert is set to silent, have not tried other lane alert options.. this error pops up intermittently on rides. 

Sometimes I am able to reengage openpilot, but most of the time the vehicle needs to be restarted in order to work again.

brittonx, 2022-03-29:
I have had the Low Memory Reboot Device error sometimes in the morning when first turning on the car.  I did not see this on stock.

Colorlessroom, 2022-05-27:
I have this issue as well; 2021 RAV4 Hybrid. After using this fork for sometime it will continue to show "low memory please reboot" until it finally fails to launch the software and shows a error message. [image]

After that I had cleared it and reinstalled the fork and "low memory" message went a while for a week or so and is now coming on every other drive almost. I figure soon it will fail to boot again.